RobotWebTools/web_video_server #5: web_video_server is not a full replacement for ros_web_video yet

psoetens, 2015-01-29:
ros_web_video had a workaround for serving 3D pointcloud data to ros3djs by also being the file server. This feature appears to be dropped from web_video_server and it's impossible to serve point clouds using it.

rctoris, 2015-01-29:
@mitchellwills is this able to do this yet?

mitchellwills, 2015-01-30:
Hmm, it looks like I missed this. For some reason I think that I had thought the cross domain issue had been solved when I was working on this. I'll take a look this weekend and if it's not functioning i'll implement it. Should be pretty simple.

psoetens, 2015-01-30:
Aside from the pull request, I think you are correct that it is no longer required, at least with recent Chrome and Firefox versions. I think my previous tests failed due to a wrong URL of the stream.
I now have a setup where the pages are served from port 80 and the depth cloud from port 9999 by web_video_server. Any info on when this issue was resolved in the past ?

mitchellwills, 2015-02-01:
I submitted a pull request (RobotWebTools/ros3djs#93) that updates the depthcloud example to use web_video_server. You don't actually need to serve the webpage from the same server anymore.

closed 2015-07-07 by rctoris