RobotWebTools/web_video_server #7: vp8 encoding is flakey in browsers and in vlc

psoetens, 2015-02-04:
Using the develop branch: We saw that after a time, the vp8 streams in a browser slowed down after 5-10 seconds to about 1 frame every few seconds. Playing the same stream with vlc version 2.1.4 gives these outputs during such slowdown:

[0x7f9ce80009b8] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
[0x7f9ce80009b8] main input error: ES_OUT_RESET_PCR called
[vp8 @ 0x7f9cd8c3fea0] Discarding interframe without a prior keyframe!
[vp8 @ 0x7f9cd8c40300] Discarding interframe without a prior keyframe!
[vp8 @ 0x7f9cd8c42c40] Discarding interframe without a prior keyframe!
[vp8 @ 0x7f9cd8c455a0] Discarding interframe without a prior keyframe!
[vp8 @ 0x7f9cd8c3fea0] Discarding interframe without a prior keyframe!
[vp8 @ 0x7f9cd8c40300] Discarding interframe without a prior keyframe!

vlc eventually recovers and the stream becomes fluent again, but our Chrome browser does not.

http://superuser.com/questions/589796/how-to-remove-interframe-errors-from-webm

This issue doesn't crop up when streaming as an mjpeg, although vlc gives in that case:
"[mjpeg @ 0x7f0168c277e0] No JPEG data found in image" a zillion times...

mitchellwills, 2015-02-05:
Do you see this in just the devel branch or also the master branch?

rctoris, 2015-02-05:
I actually saw this on the debian package as well.

mitchellwills, 2015-02-05:
I did see some of this at one point a few months ago (I haven't really used this package since) in chrome. I think that something changed in chrome potentially (and this is not a regression in the move from ros_web_video) as running the same stream side by side yields similar results in chrome. In chrome I get short clips of video followed by periods where I get a frame every few seconds. Both streams perform significantly better in firefox for some reason. I see similar results with web_video_server in vlc, but I was unable to compare it to ros_web_video as vlc complains about the headers for some reason.

As for the actual errors i'm not totally sure what is causing the problem. Nothing in your link seems to point to any clear source of the problem. The only thing that I could possibly think of would be that the specified framerate of the video doesn't match the rate at which frames are being sent. I wonder if this is causing vlc to drop frames (and occasionally one of these is an interframe).

mitchellwills, 2015-02-05:
No idea about the mjpeg error message