RobotWebTools/web_video_server #12: handling compressed image

jihoonl, 2015-03-10:
`web_video_server` is not able to subscribe compressed image topics.

jihoonl, 2015-03-10:
Never mind. It should be handled at image transport level.

closed 2015-03-10 by jihoonl

checked01, 2019-09-26:
@jihoonl I am also unable to subscribe to compressed image topics.

I've been trying to access them via:
`http://localhost:8080/stream?topic=/cameras/bw/image_raw?type=ros_compressed`
Where there is an active topic at:
`/cameras/bw/image_raw/compressed`

But no stream is resolved. Additionally on the web_video_server landing page at localhost:8080 I only see raw uncompressed image topics, despite the compressed topics also being published.

Were you able to find a way to subscribe to compressed images?

ngocpham1994x, 2022-12-17:
@swansk @jihoonl  I am currently having the same issue.
@swansk Did you figure out? Does it have anything to do with the image_transport plugins?
Appreciate any help to point me to the correct direction.
Thanks