RobotWebTools/web_video_server #13: Configurable server side video buffer size

jgoppert, 2015-05-04:
When out of range off wifi on our video server, a buffer starts building on the server side which is visible using top as the memory grows on the ros web video process. When back in range, the buffer goes back down and the video plays back in high speed. We would like the ability to limit this buffer so that we don't develop a long video lag when getting poor reception.

mitchellwills, 2015-05-08:
Hmm there is no buffering in the application so it must be TCP buffering the data. I've never tried to get information about the TCP buffer before. Unfortunately this is why TCP is inherently not good for real time data like this. I've been working on a bridge for webrtc, which provides much better streaming performance (RobotWebTools/webrtc_ros)

RobertRoe, 2015-09-14:
We recently had the same problem. The web _video_server is now using the async_web_server_cpp. There we found a buffer for pending writes which fills up during connection outages. Depending on your memory size and quality of the video stream this can result in a delayed video stream up to minutes. 

https://github.com/WPI-RAIL/async_web_server_cpp

Because lack of time we simply limited the size of the buffer by not allowing it to grow over two frames.  

async_web_server -> http_connetction.cpp --> HttpConnection::write(.....) --> pending_write_buffers and pending_write_ressources 

This worked well for us.

BennyRe, 2015-09-14:
Great news you found a cause for this problem.
Maybe you could create a pull request with a configurable buffer length? :heart_eyes_cat:

SystemDiagnosticss, 2018-02-21:
@mitchellwills Hello, please say how you install webrtc_ros ???

malcom2073, 2019-12-11:
I just ran into this issue as well, except the buffer will actually fill up to the point of using all the ram and crashing the system if you never come back to it. Recompiling a custom ros-kinetic-async-web-server-cpp to do what @RobertRoe recommended (Thanks for the suggestion!) worked great. It would be nice if this supported either customization of buffer sizes or detecting and closing dead connections, but I believe those are limitations of async-web-server-cpp, not of web_video_server.