RobotWebTools/web_video_server #14: How can I adjust the encoding performance?

nassarofficial, 2015-05-11:
How can I adjust the encoding performance of realtime vp8 encoding. The documentation specifies an algorithm, how can I edit it?

mitchellwills, 2015-05-12:
Can you be more specific on what you want to edit. web_video_server uses ffmpeg/libav to do vp8 encoding. The parameters that are exposed are described here: http://wiki.ros.org/web_video_server#VP8_Stream_Parameters
You can see more specifically how they are applied here: https://github.com/RobotWebTools/web_video_server/blob/develop/src/libav_streamer.cpp#L95 and https://github.com/RobotWebTools/web_video_server/blob/develop/src/vp8_streamer.cpp#L52

rctoris, 2015-07-07:
Closing for inactivity.

closed 2015-07-07 by rctoris