RobotWebTools/web_video_server #17: not able to catkin_make develop branch

dhirajdhule, 2015-09-22:
This is the error I get when I try to catkin_make web_video_server 
flight@flight:~/alphaws$ catkin_make
Base path: /home/flight/alphaws
Source space: /home/flight/alphaws/src
Build space: /home/flight/alphaws/build
Devel space: /home/flight/alphaws/devel
Install space: /home/flight/alphaws/install
####
#### Running command: "make cmake_check_build_system" in "/home/flight/alphaws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/flight/alphaws/build"
####
[ 14%] Building CXX object web_video_server/CMakeFiles/web_video_server.dir/src/ros_compressed_streamer.cpp.o
/home/flight/alphaws/src/web_video_server/src/ros_compressed_streamer.cpp: In member function ‘void web_video_server::RosCompressedStreamer::imageCallback(const CompressedImageConstPtr&)’:
/home/flight/alphaws/src/web_video_server/src/ros_compressed_streamer.cpp:32:90: error: no matching function for call to ‘web_video_server::MultipartStream::sendPart(const _stamp_type&, std::string&, boost::asio::const_buffers_1, const CompressedImageConstPtr&)’
     stream_.sendPart(msg->header.stamp, content_type, boost::asio::buffer(msg->data), msg);
                                                                                          ^
/home/flight/alphaws/src/web_video_server/src/ros_compressed_streamer.cpp:32:90: note: candidate is:
In file included from /home/flight/alphaws/src/web_video_server/include/web_video_server/ros_compressed_streamer.h:8:0,
                 from /home/flight/alphaws/src/web_video_server/src/ros_compressed_streamer.cpp:1:
/home/flight/alphaws/src/web_video_server/include/web_video_server/multipart_stream.h:18:8: note: void web_video_server::MultipartStream::sendPart(const ros::Time&, const string&, const boost::asio::const_buffer&, async_web_server_cpp::HttpConnection::ResourcePtr)
   void sendPart(const ros::Time &time, const std::string& type, const boost::asio::const_buffer &buffer,
        ^
/home/flight/alphaws/src/web_video_server/include/web_video_server/multipart_stream.h:18:8: note:   no known conversion for argument 4 from ‘const CompressedImageConstPtr {aka const boost::shared_ptr<const sensor_msgs::CompressedImage_<std::allocator<void> > >}’ to ‘async_web_server_cpp::HttpConnection::ResourcePtr {aka boost::shared_ptr<void>}’
make[2]: *** [web_video_server/CMakeFiles/web_video_server.dir/src/ros_compressed_streamer.cpp.o] Error 1
make[1]: *** [web_video_server/CMakeFiles/web_video_server.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

##########################

When I switched to 0.0.3 (detached head state) catkin_make works.
But not sure what functionality will not be available then.

BennyRe, 2015-09-22:
Is your async_web_server_cpp still on 0.0.2? If yes update to 0.0.3.

dhirajdhule, 2015-09-22:
It works after updating to 0.0.3.
Thanks

BennyRe, 2015-09-22:
Good to hear that it works.
Please close this issue.

closed 2015-09-22 by rctoris