RobotWebTools/web_video_server #20: Upgrade for ffmpeg 3.0?

hawesie, 2016-03-16:
web_video_server requires a version of ffmpeg < 3.0. In 3.0.0 some of the identifiers used in libav_streamer.cpp have been changed, e.g. `avcodec_alloc_frame` and `PIX_FMT_BGR24`.

BennyRe, 2016-03-17:
Hey,

dependency upgrades should be done with ROS distro upgrades. So, a new version of ffmpeg could be used with ROS Kinetic. As specified in [REP 3](http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021) the target platforms of Kinetic are Ubuntu Wily 15.10 and Xenial 16.04.

[Launchpad](https://launchpad.net/ubuntu/+source/ffmpeg) names ffmpeg 2.7 and 2.8 as the released versions of ffmpeg in Ubuntu 15.10 and 16.04. So, in my opinion web_video_server should stay < 3.0 for Kinetic.

hawesie, 2016-03-17:
Ok, that makes a lot of sense to me. 

For OS X users, Homebrew now installs 3.0, so we will need to manage dependencies manually (but we're very much a minority here!).

closed 2016-03-17 by hawesie