RobotWebTools/web_video_server #21: Topic URLs unresponsive

chrisspen, 2016-03-18:
When I run a video capture node that directly outputs mjpeg format, the web_video_server index page lists this topic:

    /camera/

which I can successfully view with:

    rosrun image_view image_view image:=/camera/image _image_transport:=compressed

However, none of the suggested web_video_server URLs in the docs work.

For example, `http://localhost:8080/stream?topic=camera&type=mjpeg` returns "Unable to connect".

Does web_video_server only work with certain types of video formats or am I using the wrong URL?

BennyRe, 2016-03-23:
Does `http://localhost:8080` show you a list of topics?

chrisspen, 2016-03-24:
No, with only a node providing /camera/image/compressed, it shows nothing. If I start a node exposing the uncompressed stream at /camera/image, then it lists that topic.

kilmarnock, 2016-05-11:
I can confirm that http://localhost:8080 does not list anything. It says "Available ROS Image Topics:", the rest is empty.

chrisspen, 2016-05-11:
@kilmarnock, If you're interested, [this](https://gist.github.com/chrisspen/8034399b8d70c6d5a530c726433802ab) is what I ended up using to stream compressed video.

kilmarnock, 2016-05-13:
@chrisspen 
Thank you for the link. I adapted it to my needs and it did its work over night without consuming memory or cpu.

closed 2018-01-08 by jihoonl

checked01, 2019-09-26:
I am experiencing the same issue with compressed image topics not being displayed on the `localhost:8080` web_video_server landing page. Compressed image topics are also just inaccessible, even specifying the `ros_compressed` type in the URL for web_video_server.

Is web_video_server simply unable to subscribe to compressed topics? If so @kilmarnock what was your solution to stream compressed video? The previously posted link seems to be broken, thanks!

Jaribeau, 2019-12-07:
I'm also keen to get this working - I'm trying to use [mjpeg_cam](https://github.com/rsmohamad/mjpeg_cam) and [web_video_server](http://wiki.ros.org/web_video_server) to remove the need for compression an unnecessary round of decoding and encoding before streaming over the network. 
@chrisspen - the link to that gist is no longer available, do you have the solution elsewhere? 
Thanks!

chrisspen, 2019-12-07:
@Jaribeau  I think the link was just [this](https://github.com/chrisspen/homebot/blob/master/src/ros/src/ros_homebot_teleop/nodes/mjpeg_streamer.py).