RobotWebTools/web_video_server #25: Segmentation Fault with yuyv Webcam

DanielArnett, 2016-06-03:
I am trying to stream a webcam using web_video_server.

I am using [usb_cam](http://wiki.ros.org/usb_cam) to publish my webcam as a rostopic. My webcam is yuyv format video. When I try to connect to the video using `http://localhost:8080/stream_viewer?topic=/usb_cam/image_raw` the streaming server faults.

```
    Handling Request: /snapshot?topic=/usb_cam/image_raw
    Segmentation fault (core dumped)
```

What am I doing wrong?

SaifAlabachi, 2016-11-29:
Hi, I'm having the same issue but with ardrone camera, so I'm interested to know if you resolve your issue and maybe it will work with mine,
Thanks,

DanielArnett, 2016-11-29:
Unfortunately I can't help you with that issue. Is this the AR drone 2.0? 

One thing I do recommend is CVDrone for working with the AR drone.
https://github.com/puku0x/cvdrone

cyberphantom, 2016-12-06:
Yes it is ardrone 2,
Thank you for the cvdrone, I'm still working on it