RobotWebTools/web_video_server #29: Release for ROS Kinetic

pjreed, 2016-11-04:
It looks like currently this project has only been released for ROS versions up through Jade.  It'd be nice if it was available in Kinetic.

jack-oquin, 2016-11-15:
Our project would benefit from a Kinetic release, too: utexas-bwi/bwi_common#83.

If I try to build it on Xenial with Kinetic, I get an unresolved rosdep for ``async_web_server_cpp``. How should that be resolved?

jack-oquin, 2016-11-15:
It looks like a Kinetic release of async_web_server_cpp is pending: GT-RAIL/async_web_server_cpp#7.

jack-oquin, 2016-11-15:
Building along with the ``async_web_server_cpp`` sources, yields a bunch of deprecation warnings for some avcodec interfaces:
```
Warnings   << web_video_server:make /home/joq/ros/ws/logs/web_video_server/build.make.000.log                                                                                  
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp: In destructor ‘web_video_server::LibavStreamer::~LibavStreamer()’:
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:75:5: warning: ‘void avcodec_free_frame(AVFrame**)’ is deprecated [-Wdeprecated-declarations]
     avcodec_free_frame(&frame_);
     ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:3952:6: note: declared here
 void avcodec_free_frame(AVFrame **frame);
      ^
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:75:5: warning: ‘void avcodec_free_frame(AVFrame**)’ is deprecated [-Wdeprecated-declarations]
     avcodec_free_frame(&frame_);
     ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:3952:6: note: declared here
 void avcodec_free_frame(AVFrame **frame);
      ^
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:75:31: warning: ‘void avcodec_free_frame(AVFrame**)’ is deprecated [-Wdeprecated-declarations]
     avcodec_free_frame(&frame_);
                               ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:3952:6: note: declared here
 void avcodec_free_frame(AVFrame **frame);
      ^
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp: In member function ‘virtual void web_video_server::LibavStreamer::initialize(const cv::Mat&)’:
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:177:12: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated [-Wdeprecated-declarations]
   frame_ = avcodec_alloc_frame();
            ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:3927:10: note: declared here
 AVFrame *avcodec_alloc_frame(void);
          ^
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:177:12: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated [-Wdeprecated-declarations]
   frame_ = avcodec_alloc_frame();
            ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:3927:10: note: declared here
 AVFrame *avcodec_alloc_frame(void);
          ^
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:177:32: warning: ‘AVFrame* avcodec_alloc_frame()’ is deprecated [-Wdeprecated-declarations]
   frame_ = avcodec_alloc_frame();
                                ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:3927:10: note: declared here
 AVFrame *avcodec_alloc_frame(void);
          ^
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp: In member function ‘virtual void web_video_server::LibavStreamer::sendImage(const cv::Mat&, const ros::Time&)’:
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:294:25: warning: ‘AVCodecContext::coded_frame’ is deprecated [-Wdeprecated-declarations]
     if (codec_context_->coded_frame->key_frame)
                         ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:3024:35: note: declared here
     attribute_deprecated AVFrame *coded_frame;
                                   ^
/home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:294:25: warning: ‘AVCodecContext::coded_frame’ is deprecated [-Wdeprecated-declarations]
     if (codec_context_->coded_frame->key_frame)
                         ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:3024:35: note: declared here
     attribute_deprecated AVFrame *coded_frame;
                                   ^
[... 14 lines cut ...]
                 from /home/joq/ros/ws/src/web_video_server/src/vp8_streamer.cpp:37:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:2618:11: note: declared here
     float rc_buffer_aggressivity;
           ^
[... cut ...]
   codec_context_->rc_buffer_aggressivity = 0.5;
                   ^
In file included from /home/joq/ros/ws/src/web_video_server/include/web_video_server/libav_streamer.h:11:0,
                 from /home/joq/ros/ws/src/web_video_server/include/web_video_server/vp8_streamer.h:41,
                 from /home/joq/ros/ws/src/web_video_server/src/vp8_streamer.cpp:37:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:2618:11: note: declared here
     float rc_buffer_aggressivity;
           ^
```

It does build, however.

jihoonl, 2016-11-16:
Waiting for https://github.com/GT-RAIL/async_web_server_cpp/issues/7 to be release. After all dependencies are ready, I will try to release this repo.

jack-oquin, 2016-12-21:
``async_web_server`` was just released, ref: GT-RAIL/async_web_server_cpp#7

jihoonl, 2017-01-17:
should be available in next cycle.

closed 2017-01-17 by jihoonl