RobotWebTools/web_video_server #34: Failed to run web video server

jihoonl, 2017-04-03:
```
> rosrun web_video_server web_video_server 
/opt/ros/kinetic/lib/web_video_server/web_video_server: error while loading shared libraries: libopencv_imgproc3.so.3.1: cannot open shared object file: No such file or directory
```

pedro1713, 2017-04-03:
Works OK on two different computers for me

SimonBirrell, 2017-04-10:
Similar:

    rosrun web_video_server web_video_server

/opt/ros/kinetic/lib/web_video_server/web_video_server: error while loading shared libraries: libopencv_core3.so.3.2: cannot open shared object file: No such file or directory

jihoonl, 2017-04-13:
It looks like the version change in [cv_bridge](https://github.com/ros-perception/vision_opencv/tree/kinetic/cv_bridge) caused the problem. I have confirmed that `web_video_server` works with cv_bridge version 1.11.15

jihoonl, 2017-08-30:
It was library linking issue in my pc. I have confirmed that no problem occurred after fixing library dependency in my pc.

closed 2017-08-30 by jihoonl

vannem95, 2018-03-28:
@jihoonl How do i fix the cv_bridge library dependency? I am getting the same error.