RobotWebTools/web_video_server #56: Black Image of Depth Image

Maverobot, 2018-01-29:
Hallo,

since last week the depth image streaming is not working any more and showing only black images, while the rgb image is still functional. I tried this on multiple PCs. Same happened. Can this be confirmed by you developers?  

Cheers

jihoonl, 2018-01-30:
#52 and #53 merged in since last week. do you remember which version you were using before? And do you use depthcloud_encoder too?

Maverobot, 2018-01-30:
Sorry, I don't remember which version I was using before. But I think I tried both develop and master branch. Neither worked. And I just tested the latest develop branch. It is still delivering only black depth images.

PS: from Rviz I could visualize the depth image. No problem there.

jihoonl, 2018-02-08:
I have confirmed that web_video_server streams depth image fine. It is a lot darker then the rviz view though.

[image]

Maverobot, 2018-02-10:
@jihoonl Thanks for the confirmation. Is it intentional or is it still in the plan to visualize it a bit more readable? :)

closed 2024-10-01 by bjsowa