RobotWebTools/web_video_server #57: Build failing on Debian Jessie Kinetic

tfoote, 2018-01-31:
The builds are failing on Debian Jessie it looks to be related to calling an FFMPEG API call that is possibly different on Jessie. This looks like a similar issue to #47 

```
[ 42%] Building CXX object CMakeFiles/web_video_server.dir/src/libav_streamer.cpp.o
/usr/lib/ccache/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"web_video_server\" -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -D_FORTIFY_SOURCE=2  -I/tmp/binarydeb/ros-kinetic-web-video-server-0.0.7/include -I/opt/ros/kinetic/include -I/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /opt/ros/kinetic/include/opencv-3.3.1 -isystem /opt/ros/kinetic/include/opencv-3.3.1/opencv    -o CMakeFiles/web_video_server.dir/src/libav_streamer.cpp.o -c /tmp/binarydeb/ros-kinetic-web-video-server-0.0.7/src/libav_streamer.cpp
/tmp/binarydeb/ros-kinetic-web-video-server-0.0.7/src/libav_streamer.cpp: In member function ‘virtual void web_video_server::LibavStreamer::sendImage(const cv::Mat&, const ros::Time&)’:
/tmp/binarydeb/ros-kinetic-web-video-server-0.0.7/src/libav_streamer.cpp:229:87: error: ‘av_image_fill_arrays’ was not declared in this scope
                        img.data, input_coding_format, output_width_, output_height_, 0);
```
It's failing for both arm64 and amd64 Jessie
http://build.ros.org/view/Kbin_djv8_dJv8/job/Kbin_djv8_dJv8__web_video_server__debian_jessie_arm64__binary
http://build.ros.org/view/Kbin_dj_dJ64/job/Kbin_dj_dJ64__web_video_server__debian_jessie_amd64__binary/

tfoote, 2018-01-31:
I've opened a PR to blacklist these targets to stop the failing builds until this is fixed. Once fixed https://github.com/ros-infrastructure/ros_buildfarm_config/pull/105 can be reverted.

techping, 2018-03-17:
@tfoote How can I fix it in an environment where ROS is already installed?

techping, 2018-03-17:
Use tag 0.0.6(```git checkout 0.0.6```) can build successfully.

closed 2024-10-01 by bjsowa