RobotWebTools/web_video_server #59: Segmentation fault when requesting libav stream of non-published topic

jabailey, 2018-02-21:
To reproduce: run `web_video_server` like normal, then point browser to `http://localhost:8080/stream_viewer?topic=/some/unpublished/topic&type=vp8`

Results:

```
root@master:/catkin_ws# rosrun web_video_server web_video_server
[ INFO] [1519240063.734495445]: Waiting For connections on 0.0.0.0:8080
[ INFO] [1519240067.188024505]: Handling Request: /stream_viewer?topic=/camera/image_rect_color&type=vp8
[ INFO] [1519240067.303481877]: Handling Request: /stream?topic=/camera/image_rect_color&type=vp8
[ INFO] [1519240067.732592846]: Removed Stream: /some/unpublished/topic
Segmentation fault (core dumped)
```

The segfault seems to stem from referencing `format_context_` in the `LibavStreamer` destructor before it's been allocated: https://github.com/RobotWebTools/web_video_server/blob/develop/src/libav_streamer.cpp#L80

Quick fix (tested on a local branch):

```diff
diff --git a/src/libav_streamer.cpp b/src/libav_streamer.cpp
index 08980e4..8682667 100644
--- a/src/libav_streamer.cpp
+++ b/src/libav_streamer.cpp
@@ -77,10 +77,11 @@ LibavStreamer::~LibavStreamer()
   }
   if (io_buffer_)
     delete io_buffer_;
-  if (format_context_->pb)
-    av_free(format_context_->pb);
-  if (format_context_)
+  if (format_context_) {
+    if (format_context_->pb)
+      av_free(format_context_->pb);
     avformat_free_context(format_context_);
+  }
   if (sws_context_)
     sws_freeContext(sws_context_);
 }
```

SystemDiagnosticss, 2018-02-22:
@jabailey Please say, after your fix this bug your videostreaming (with codec vp8) has big latency or not?? Can show it(in youtube or anywhere )?

jihoonl, 2018-02-24:
Patch looks good to me. could you create a PR to patch this?

jabailey, 2018-02-26:
@jihoonl https://github.com/RobotWebTools/web_video_server/pull/60

jabailey, 2018-02-26:
@SystemDiagnosticss Yes, I've found that there is high latency - the stream takes about 5-10 seconds to initialize, and the frames play 5-10 seconds behind. I don't see how my change could affect that, but I'll admit I'm not very familiar with this code. Is the latency lower without my change?

jihoonl, 2018-02-26:
I saw the high latency too with codec vp8. @randoms do you have any idea?

closed 2018-02-26 by jihoonl

SystemDiagnosticss, 2018-02-26:
@jabailey @jihoonl  In case mjpeg stream I have 2-3 sec latency. But its very very big latency. I am new in ROS but situation with videostream is very funny for me. I work with videostream without ROS (Raspberry Pi to web page) and has latency 0.2 sec. Its uv4l libs for videostream. https://www.linux-projects.org/uv4l/tutorials/  This project use webrtc. I try use webrts for ROS (http://wiki.ros.org/webrtc_ros) but I can't install it. I create questions on rosanswer, stackoverflow but not receive an answer. I don't understand is it really in ROS does not have good videostreaming solution ??  This is very strange

commit 34415273 (2018-02-26): Fix segfault in libav_streamer destructor (resolves #59) (#60)