RobotWebTools/web_video_server #61: web_video_server crash every time I run ROS node which publish camera/image msg, on Ubuntu 14.04 + ROS Indigo

wqdun, 2018-03-30:
The video server core dump when I start camera node, which publish camera/image type message.
Then I realized I source my own setup.bash before I run web_video_server, I changed my code to
"
     . /opt/ros/indigo/setup.bash
    /opt/ros/indigo/lib/web_video_server/web_video_server >>$result_log 2>&1 &
"
and it works.
But I still do not know why, wonder can it be called a bug?
Seems we need different setup files to run rosbridge and web_video like below:
"
    . /opt/smartc/devel/setup.bash  # this is my own setup
    roslaunch /opt/ros/indigo/share/rosbridge_server/launch/rosbridge_websocket.launch >>$result_log 2>&1 &
    . /opt/ros/indigo/setup.bash
    /opt/ros/indigo/lib/web_video_server/web_video_server >>$result_log 2>&1 &
"

ysh11o, 2019-05-25:
I have same problem with you.
try $ sudo apt-get install web_video_server

ysh11o, 2019-05-25:
then run $ catkin make

closed 2024-10-01 by bjsowa