RobotWebTools/web_video_server #72: web_video_server wont run

letsdoitagain, 2018-09-09:
Hello,
upon entering:   **rosrun web_video_server web_video_server**

i get the following error: 
```
[ERROR] [1536532836.635104100]: Exception when creating the web server! 0.0.0.0:8080
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  bind: Address already in use
Aborted (core dumped)
```
I am using ros kinetic. Not sure how to fix it.

Best,

kenny-y, 2018-09-10:
It looks like that port was already taken by another program: `Address already in use`

letsdoitagain, 2018-09-10:
nope; 

`sudo netstat -ntlp | grep :8080`

gives

`tcp        0      0 127.0.0.1:8080`

so; it is listening.