RobotWebTools/web_video_server #82: MultipartStream not leaving busy state (leading to high CPU usage)

sfalexrog, 2019-01-24:
ROS distro: kinetic
Package version: 0.1.0

I've noticed that sometimes, even after I close all tabs with web_video_server's streams, the web_video_server's process still consumes as much CPU as if it was still streaming. After a bit of debugging, I've found that [```bool Multipart::isBusy()```](https://github.com/RobotWebTools/web_video_server/blob/d2886061b8ce078370023020eac32bee060facb0/src/multipart_stream.cpp#L62-L67) always returns true, unable to clean up ```pending_footers_```. This prevents ```MultipartStreamer``` from sending additional packets (and, from what I can tell, getting an exception and setting ```ImageStreamer::inactive_``` flag).

This happens (and is most noticeable) on a Raspberry Pi, not so much on a desktop - sometimes I have to close and reopen the tabs multiple times, in quick succession.

I've attempted to work around this by having a timestamp assigned to every pending footer and removing "stale" footers: https://github.com/sfalexrog/web_video_server/commit/115f906bb90a2354e6d79518daf5920687701966. This is by no means a proper fix, though; any comments regarding the issue are very welcome!

130s, 2019-06-11:
https://github.com/RobotWebTools/web_video_server/pull/83 seems to have addressed this already?

sfalexrog, 2019-06-11:
Well, that's more of a workaround than of a proper fix, but the problem doesn't seem to manifest itself anymore. Guess the issue could be closed now.

closed 2019-06-11 by sfalexrog