RobotWebTools/web_video_server #85: undefined reference to `avcodec_send_frame'

codezealot, 2019-02-13:
when compile web_video_server,

the system report error:
[100%] Linking CXX executable /edge_computing/ros/ros/devel_isolated/web_video_server/lib/web_video_server/web_video_server
CMakeFiles/web_video_server.dir/src/libav_streamer.cpp.o: In function `web_video_server::LibavStreamer::sendImage(cv::Mat const&, ros::Time const&)':
libav_streamer.cpp:(.text+0x970): undefined reference to `avcodec_send_frame'
libav_streamer.cpp:(.text+0x98b): undefined reference to `avcodec_receive_packet'
collect2: error: ld returned 1 exit status

What am I doing wrong?

sneheshs, 2021-06-15:
Did you ever figure out what the issue was? Mine was working fine a few years ago. I tried recompiling and got the same error now.

sneheshs, 2024-10-01:
Closed? Why?

bjsowa, 2024-10-01:
> Closed? Why?

This issue seems to be related to using old ffmpeg libraries on EoL Ubuntu distros.  I can reopen it if you believe it still applies to currently supported versions (at least Ubuntu 20.04, ffmpeg 4.2.7). 

Otherwise, I would suggest git bisecting the repo to find the last working commit which will probably be 3f7ffeb as the `avcodec_send_frame` and `avcodec_receive_packet` functions were added in a9dff84

closed 2024-10-01 by bjsowa

sneheshs, 2024-10-02:
I see, ok thanks for the clarification. Yeah we can close it then.