RobotWebTools/web_video_server #94: Compressed image topic not accessible

checked01, 2019-09-17:
I have two compressed image topics that I am publishing. For example, one is at `/robot/cameras/bw/image_raw/compressed`. I verified that the topic is subscribed to by `/web_video_server` by checking `rqt_graph`. 

When I visit http://localhost:8080, I can see the two camera topics listed without any links to click on to navigate to the stream or image snapshots.

Both image topics have camera_info additionally being published, and if I publish the raw_image without compression it seems to work fine.

MoffKalast, 2020-04-07:
Still seems to be an issue at this time. I have a `CompressedImage `stream at `/raspicam_node/image/compressed` yet no topics are listed at localhost and trying to access it at say:

`http://localhost:8080/stream?topic=/raspicam_node/image&type=ros_compressed`

or 

`http://localhost:8080/stream?topic=/raspicam_node/image/compressed&type=ros_compressed`

gives nothing.

But given that most of the documentation seems to focus on compression instructions I guess this server doesn't even support publishing pre-compressed messages and merely accepts raw to compress and send. Kind of a bummer though.

Doolly, 2021-03-04:
any updates on this issue? I'm having the same issue

ngocpham1994x, 2022-12-17:
I am currently having same issue.

bjsowa, 2024-10-07:
With the way it's currently implemented, compressed topics won't be listed in stream list, but they still should be accessible if using `ros_compressed` type stream. 

For example, if the camera driver is publishing `/camera/image_raw/compressed` topic, it can be accessed through this URL:
```
http://127.0.0.1:8080/stream?topic=/camera/image_raw&type=ros_compressed
```

closed 2025-12-12 by bjsowa