RobotWebTools/web_video_server #101: catkin_make fails on jetson nano

NikolaeVarius, 2020-05-19:
`-- ==> add_subdirectory(web_video_server)
-- Could NOT find async_web_server_cpp (missing: async_web_server_cpp_DIR)
-- Could not find the required component 'async_web_server_cpp'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "async_web_server_cpp" with any of the following names:

    async_web_server_cppConfig.cmake
    async_web_server_cpp-config.cmake

  Add the installation prefix of "async_web_server_cpp" to CMAKE_PREFIX_PATH
  or set "async_web_server_cpp_DIR" to a directory containing one of the
  above files.  If "async_web_server_cpp" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  web_video_server/CMakeLists.txt:7 (find_package)
`
Ubuntu 18.04 on a Jetson nano

chenkaiC4, 2020-07-05:
you can try install ros-melodic-web-video-server, then build this repo.

peterlixue, 2020-08-11:
1.  sudo apt-get install ros-melodic-web-video-server;
2.  if find some error, then sudo apt-get update, again , apt install ros-melodic-web-video-server
3.  use ros catkin_make it,  show target generation OK

closed 2024-10-01 by bjsowa