RobotWebTools/web_video_server #102: Noetic release

jspricke, 2020-06-03:
Please release this into the Noetic distribution.

jihoonl, 2020-06-20:
Same here. Could you help me to add travis config for noetic?

Tobias-Fischer, 2020-09-21:
Same here :+1:

jihoonl, 2020-09-23:
This needs to be released first..
https://github.com/GT-RAIL/async_web_server_cpp/issues/18

Carter12s, 2020-10-16:
Need this on noetic as well.

Attempted building myself, and everything seems to compile.

Able to run the web_video_server node.

mjpeg streams work, but no video streams work.

Logs with debug enabled:

`
[DEBUG] [1602809608.753363088] [/web_video_server] Library lib/libimage_transport_plugins found at explicit path /opt/ros/noetic/lib//libimage_transport_plugins.so.
[DEBUG] [1602809608.753389218] [/web_video_server] image_transport/raw_sub maps to real class type image_transport::RawSubscriber
[DEBUG] [1602809608.753415248] [/web_video_server] boost::shared_ptr to object of real type image_transport::RawSubscriber created.
[DEBUG] [1602809608.755195417] [/web_video_server] Publisher update for [/visualization/object_analysis/image]: http://localhost:42795/,  already have these connections: 
[DEBUG] [1602809608.755357667] [/web_video_server] Began asynchronous xmlrpc connection to [localhost:42795]
[DEBUG] [1602809608.892787686] [/web_video_server] Connecting via tcpros to topic [/visualization/object_analysis/image] at host [localhost:51453]
[DEBUG] [1602809608.893240999] [/web_video_server] Resolved publisher host [localhost] to [127.0.0.1] for socket [30]
[DEBUG] [1602809608.893332609] [/web_video_server] Adding tcp socket [30] to pollset
[DEBUG] [1602809608.893356999] [/web_video_server] Async connect() in progress to [localhost:51453] on socket [30]
[DEBUG] [1602809608.893408469] [/web_video_server] Async socket[30] is connected
[DEBUG] [1602809608.893452249] [/web_video_server] Connected to publisher of topic [/visualization/object_analysis/image] at [localhost:51453]
[DEBUG] [1602809610.394310812] [/web_video_server] system_error exception: Broken pipe
[ INFO] [1602809610.859152481] [/web_video_server] Removed Stream: /visualization/object_analysis/image
[DEBUG] [1602809610.859198671] [/web_video_server] Destroying ClassLoader, base = image_transport::PublisherPlugin, address = 0x7f04581e6e00
[DEBUG] [1602809610.859936594] [/web_video_server] Connection::drop(2)
[DEBUG] [1602809610.859981115] [/web_video_server] TCP socket [30] closed
[DEBUG] [1602809610.859996235] [/web_video_server] Connection::drop(0)
[DEBUG] [1602809610.860012645] [/web_video_server] Connection::drop(2)
[DEBUG] [1602809610.860026935] [/web_video_server] Connection::drop(2)
[DEBUG] [1602809610.860054045] [/web_video_server] Subscriber on '/visualization/object_analysis/image' deregistering callbacks.
[DEBUG] [1602809610.860069045] [/web_video_server] Destroying ClassLoader, base = image_transport::SubscriberPlugin, address = 0x7f04581cffd0
`

rereee3, 2020-12-16:
> Need this on noetic as well.
>
> Attempted building myself, and everything seems to compile.
>
> Able to run the web_video_server node.
>
> mjpeg streams work, but no video streams work.
>
> Logs with debug [...]

I've tested this myself using [sfalexrog's branch of the async_web_server_cpp](https://github.com/GT-RAIL/async_web_server_cpp/pull/19) and I confirmed that mjpeg and vp8 are working just fine. Unfortunately, sfalexrog's PR is still waiting for being merged.

julianraheema, 2021-01-21:
Please release this into the Noetic distribution.

SystemDiagnosticss, 2021-02-15:
Any updates in this case?

salsicha, 2021-02-24:
Is this a drop in replacement?

https://github.com/fkie/async_web_server_cpp

Nurgak, 2021-03-04:
[async_web_server_cpp](https://github.com/fkie/async_web_server_cpp) is available for Noetic, could this package be released for Noetic as well now?

bbferka, 2021-03-30:
Can confirm @Nurgak's comment; Just compiled the package from source today; all dependencies are available as debians packages.

carlosjoserg, 2021-07-22:
Confirming building and running successfully in `noetic`.

Adding a +1 for `noetic` release.

Thanks.

jihoonl, 2021-07-23:
https://github.com/ros/rosdistro/pull/30315

jspricke, 2021-12-27:
Package was published, so closing.

closed 2021-12-27 by jspricke