RobotWebTools/web_video_server #105: ROS2 foxy build fails

nplan, 2020-06-29:
Building for ros2 foxy fails with error:
```
/home/pi/dev_ws/src/web_video_server/src/ros_compressed_streamer.cpp:22:103: error: no matching function for call to ‘rclcpp::Node::create_subscription<sensor_msgs::msg::CompressedImage>(std::__cxx11::string&, std::_Bind_helper<false, void (web_video_server::RosCompressedStreamer::*)(std::shared_ptr<const sensor_msgs::msg::CompressedImage_<std::allocator<void> > >), web_video_server::RosCompressedStreamer*, const std::_Placeholder<1>&>::type, int)’
     compressed_topic, std::bind(&RosCompressedStreamer::imageCallback, this, std::placeholders::_1), 1);
                                                                                                       ^
In file included from /home/pi/ros2_foxy/install/rclcpp/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /home/pi/ros2_foxy/install/rclcpp/include/rclcpp/executors.hpp:22,
                 from /home/pi/ros2_foxy/install/rclcpp/include/rclcpp/rclcpp.hpp:146,
                 from /home/pi/dev_ws/src/web_video_server/include/web_video_server/image_streamer.h:4,
                 from /home/pi/dev_ws/src/web_video_server/include/web_video_server/ros_compressed_streamer.h:5,
                 from /home/pi/dev_ws/src/web_video_server/src/ros_compressed_streamer.cpp:1:
/home/pi/ros2_foxy/install/rclcpp/include/rclcpp/node.hpp:211:3: note: candidate: ‘template<class MessageT, class CallbackT, class AllocatorT, class CallbackMessageT, class SubscriptionT, class MessageMemoryStrategyT> std::shared_ptr<SubscriptionT> rclcpp::Node::create_subscription(const string&, const rclcpp::QoS&, CallbackT&&, const rclcpp::SubscriptionOptionsWithAllocator<AllocatorT>&, typename MessageMemoryStrategyT::SharedPtr)’
   create_subscription(
   ^~~~~~~~~~~~~~~~~~~
/home/pi/ros2_foxy/install/rclcpp/include/rclcpp/node.hpp:211:3: note:   template argument deduction/substitution failed:
/home/pi/dev_ws/src/web_video_server/src/ros_compressed_streamer.cpp:22:32: note:   cannot convert ‘std::bind(_Func&&, _BoundArgs&& ...) [with _Func = void (web_video_server::RosCompressedStreamer::*)(std::shared_ptr<const sensor_msgs::msg::CompressedImage_<std::allocator<void> > >); _BoundArgs = {web_video_server::RosCompressedStreamer*, const std::_Placeholder<1>&}; typename std::_Bind_helper<std::__is_socketlike<_Func>::value, _Func, _BoundArgs ...>::type = std::_Bind<void (web_video_server::RosCompressedStreamer::*(web_video_server::RosCompressedStreamer*, std::_Placeholder<1>))(std::shared_ptr<const sensor_msgs::msg::CompressedImage_<std::allocator<void> > >)>](((web_video_server::RosCompressedStreamer*)this), std::placeholders::_1)’ (type ‘std::_Bind_helper<false, void (web_video_server::RosCompressedStreamer::*)(std::shared_ptr<const sensor_msgs::msg::CompressedImage_<std::allocator<void> > >), web_video_server::RosCompressedStreamer*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (web_video_server::RosCompressedStreamer::*(web_video_server::RosCompressedStreamer*, std::_Placeholder<1>))(std::shared_ptr<const sensor_msgs::msg::CompressedImage_<std::allocator<void> > >)>’}) to type ‘const rclcpp::QoS&’
     compressed_topic, std::bind(&RosCompressedStreamer::imageCallback, this, std::placeholders::_1), 1);
```

Platform: Rasbperry Pi 4 with Raspbian Buster

AviKenz, 2020-08-01:
im having the same issue. did you have any solution ?
my configuration:
- ubuntu 18.05
- ROS: dashing

nplan, 2020-08-02:
I opened a pull request #107 that fixes the build. There are still some issues though:

- `web_video_server` package does not become available to `ros2 run` after running `source dev_ws/isntall.setup.bash`. I must run `source dev_ws/install/web_video_server/share/web_video_server/local_setup.bash` instead
- `mjpeg` and `png` are the only stream types that work

DomenicP, 2020-11-20:
I tried #107 a few days ago and while that solved some of the build issues under Foxy, new ones seem to have popped up in the meantime. I opened #111 with the full set of changes I needed to get the package to work under Foxy. I also fixed some deprecation warnings along the way. I confirmed that at least the mjpeg and ros_compressed stream types are working.

sirSmart, 2021-03-17:
create_subscription(compressed_topic, std::bind(&RosCompressedStreamer::imageCallback, this, std::placeholders::_1), 1);
==>
rclcpp::QoS qos(1);
create_subscription(compressed_topic, qos, std::bind(&RosCompressedStreamer::imageCallback, this, std::placeholders::_1));

BRs

jgvictores, 2023-01-23:
Currently able to compile (minimal warnings) from source (sorry if the issue looks stale, but Foxy itself is not so old ^^).

Compiled via (use `sudo` for `apt` commands as applicable, not my case because in a docker with root; also adapt your workspace (*_ws) path):

```bash
apt update
apt install ros-foxy-async-web-server-cpp ros-foxy-cv-bridge ros-foxy-image-transport
cd /demo_ws/src
git clone --branch ros2 https://github.com/RobotWebTools/web_video_server
cd /demo_ws
colcon build --packages-select web_video_server
```

The above was done with:
- branch `ros2` at https://github.com/RobotWebTools/web_video_server/tree/c23b9078b614abdde9270ddd5d9f13c1fc4ad172
- `ros-foxy-async-web-server-cpp` (2.0.0-1focal.20221012.214333)
- `ros-foxy-cv-bridge` (3.0.7-1focal.20221021.145612)
- `ros-foxy-image-transport` (2.4.0-1focal.20221013.000257)

Run via (again, also adapt your workspace (*_ws) path):

```bash
source /demo_ws/install/setup.sh
ros2 run web_video_server web_video_server
```

PS: This is just some minimal documentation of working versions. Thanks a lot to all the effort of the developers @DomenicP and a long "et al" at https://github.com/RobotWebTools/web_video_server/pull/111

bjsowa, 2024-10-16:
We dropped support for Foxy as it is EoL (you can still build verion `1.0.1` of this package though). The new version was released for all currently supported distros (Humble, Iron, Jazzy, Rolling).

closed 2024-10-16 by bjsowa