RobotWebTools/web_video_server #108: HOW TO - ROS2

AviKenz, 2020-08-02:
I know this can sound useless. but i am sure its not when you are newbie in ROS2.
i struggle 2 days on this. and i would like to help others.
sorry for my english im not a native speaker lol.

so lets go :)

**HOW TO INSTALL**

1. open your terminal (i told you this tuto was for newbie lol)
2. run: `sudo apt-get install ros-dashing-web-video-server` to update your package tool
3. run: `sudo apt-get install ros-dashing-web-video-server` to install the library (you must type 'Y' at some point)
4. run: 'ros2 pkg list | grep web_video_server' to check if it was installed. 'web_video_server' should be outputed.

**tips:**
step 1: very important. :)
step 3: if you cant see 'web_video_server', source ros workspace by running in terminal(for ros2-dashing): `source /opt/ros/dashing/setup.bash
`

**HOW TO TEST**
1. open your terminal ( again haha)
2. run: `ros2 run web_video_server web_video_server` to start the server
3. open another terminal and...
3. start the node which output the camera stream for e.g.: `ros2 run image_tools cam2image` (image_tools is builtin in ros2-dashing. he output the webcam stream on topic 'image')
4. open a web browser and navigate to url: _http://0.0.0.0:8080/stream?topic=/image_ where '/image' is the image topic.

you can find more informations here: https://msadowski.github.io/ros-web-tutorial-pt3-web_video_server/

so thats all. no need of 'colcon build' or something like that when you just want to use the library.

duwke, 2021-06-21:
I'm guessing step 2 was actually supposed to be `sudo apt update`

vkuehn, 2021-06-22:
there is no web-video-server in foxy ?

jgvictores, 2023-01-23:
> there is no web-video-server in foxy ?

Above https://github.com/RobotWebTools/web_video_server/issues/108#issue-671608565 is okay for dashing, but for foxy, currently from source, check: https://github.com/RobotWebTools/web_video_server/issues/105#issuecomment-1400439009

bjsowa, 2025-04-23:
The package is available in ROS 2 Humble and newer distributions. Instructions on how to install it or build from source are available in README.

closed 2025-04-23 by bjsowa