RobotWebTools/web_video_server #110: Any plans for using it in ROS foxy?

alexkutsan, 2020-11-11:
Cant find ros foxy package. 
Cant; build for ros foxy because ros foxy use colcon instead of catkin. 
Converted using script : https://github.com/bponsler/ros2-support/blob/master/scripts/catkin-to-ament.py
Find out that there is dependency to 
https://github.com/GT-RAIL/async_web_server_cpp
And it is also do not support colcon. 
Converted one more time (with deleting tests)
Compiled and installed async_web_server_cpp.

But Compilation of the web_video_server  does not work: 
```[  9%] Building CXX object CMakeFiles/web_video_server.dir/src/web_video_server.cpp.o
/home/developer/ros2_ws/web_video_server/src/web_video_server.cpp:5:10: fatal error: sensor_msgs/image_encodings.h: No such file or directory
    5 | #include <sensor_msgs/image_encodings.h>
      |          ^~~~
```

botamochi6277, 2020-11-21:
**async_web_server_cpp** for ROS2 is available.  
I tried this and catch other error messages.
I use Raspberry Pi 4 and foxy.

clone packages:
```
cd ros2_workspace/src
git clone https://github.com/GT-RAIL/async_web_server_cpp.git -b ros2
git clone https://github.com/RobotWebTools/web_video_server.git -b ros2
```

install dependencies:  
```
cd ros2_workspace
rosdep install -i --from-path src --rosdistro foxy -y
```

build async_web_server_cpp: 
```
colcon build --packages-select async_web_server_cpp
```

try to build web_video_server and catch error...
```
colcon build --packages-select web_video_server
```

```
Starting >>> web_video_server
--- stderr: web_video_server                         
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find FastRTPS (missing: FastRTPS_INCLUDE_DIR FastRTPS_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /opt/ros/foxy/share/fastrtps_cmake_module/cmake/Modules/FindFastRTPS.cmake:95 (find_package_handle_standard_args)
  /opt/ros/foxy/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_c-extras.cmake:7 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_runtime/cmake/rosidl_default_runtime-extras.cmake:12 (find_package)
  /opt/ros/foxy/share/rosidl_default_runtime/cmake/rosidl_default_runtimeConfig.cmake:41 (include)
  /opt/ros/foxy/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  /opt/ros/foxy/share/rcl_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rcl_interfaces/cmake/rcl_interfacesConfig.cmake:41 (include)
  /opt/ros/foxy/share/rcl/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rcl/cmake/rclConfig.cmake:41 (include)
  /opt/ros/foxy/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /opt/ros/foxy/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  /home/ubuntu/dev_ws/install/image_transport/share/image_transport/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/ubuntu/dev_ws/install/image_transport/share/image_transport/cmake/image_transportConfig.cmake:41 (include)
  CMakeLists.txt:8 (find_package)

---
Failed   <<< web_video_server [4.39s, exited with code 1]

Summary: 0 packages finished [9.58s]
  1 package failed: web_video_server
  1 package had stderr output: web_video_server
```

botamochi6277, 2020-12-06:
I add cmake-path for Fast RTPS with `export CMAKE_PREFIX_PATH=$AMENT_PREFIX_PATH`.
And I build again and catch other errors.

```
colcon build --packages-select web_video_server
Starting >>> web_video_server
--- stderr: web_video_server                               
In file included from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/png_streamers.h:4,
                 from /home/ubuntu/dev_ws/src/web_video_server/src/png_streamers.cpp:1:
/home/ubuntu/dev_ws/install/image_transport/include/image_transport/image_transport.h:41:89: note: #pragma message: Warning: This header is deprecated. Use 'image_transport.hpp' instead
   41 | #pragma message ("Warning: This header is deprecated. Use 'image_transport.hpp' instead")
      |                                                                                         ^
In file included from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/jpeg_streamers.h:4,
                 from /home/ubuntu/dev_ws/src/web_video_server/src/jpeg_streamers.cpp:1:
/home/ubuntu/dev_ws/install/image_transport/include/image_transport/image_transport.h:41:89: note: #pragma message: Warning: This header is deprecated. Use 'image_transport.hpp' instead
   41 | #pragma message ("Warning: This header is deprecated. Use 'image_transport.hpp' instead")
      |                                                                                         ^
In file included from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/image_streamer.h:5,
                 from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/ros_compressed_streamer.h:5,
                 from /home/ubuntu/dev_ws/src/web_video_server/src/ros_compressed_streamer.cpp:1:
/home/ubuntu/dev_ws/install/image_transport/include/image_transport/image_transport.h:41:89: note: #pragma message: Warning: This header is deprecated. Use 'image_transport.hpp' instead
   41 | This header is deprecated. Use 'image_transport.hpp' instead")
      |                                                              ^

In file included from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/image_streamer.h:6,
                 from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/png_streamers.h:5,
                 from /home/ubuntu/dev_ws/src/web_video_server/src/png_streamers.cpp:1:
/home/ubuntu/dev_ws/install/image_transport/include/image_transport/transport_hints.h:41:89: note: #pragma message: Warning: This header is deprecated. Use 'transport_hints.hpp' instead
   41 | #pragma message ("Warning: This header is deprecated. Use 'transport_hints.hpp' instead")
      |                                                                                         ^
In file included from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/image_streamer.h:6,
                 from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/ros_compressed_streamer.h:5,
                 from /home/ubuntu/dev_ws/src/web_video_server/src/ros_compressed_streamer.cpp:1:
/home/ubuntu/dev_ws/install/image_transport/include/image_transport/transport_hints.h:41:89: note: #pragma message: Warning: This header is deprecated. Use 'transport_hints.hpp' instead
   41 | This header is deprecated. Use 'transport_hints.hpp' instead")
      |                                                              ^

In file included from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/image_streamer.h:6,
                 from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/jpeg_streamers.h:5,
                 from /home/ubuntu/dev_ws/src/web_video_server/src/jpeg_streamers.cpp:1:
/home/ubuntu/dev_ws/install/image_transport/include/image_transport/transport_hints.h:41:89: note: #pragma message: Warning: This header is deprecated. Use 'transport_hints.hpp' instead
   41 | #pragma message ("Warning: This header is deprecated. Use 'transport_hints.hpp' instead")
      |                                                                                         ^
/home/ubuntu/dev_ws/src/web_video_server/src/ros_compressed_streamer.cpp: In member function ‘virtual void web_video_server::RosCompressedStreamer::start()’:
/home/ubuntu/dev_ws/src/web_video_server/src/ros_compressed_streamer.cpp:22:103 error: no matching function for call to ‘rclcpp::Node::create_subscription<sensor_msgs::msg::CompressedImage>(std::string&, std::_Bind_helper<false, void (web_video_server::RosCompressedStreamer::*)(std::shared_ptr<const sensor_msgs::msg::CompressedImage_<std::allocator<void> > >), web_video_server::RosCompressedStreamer*, const std::_Placeholder<1>&>::type, int)’
   22 | essedStreamer::imageCallback, this, std::placeholders::_1), 1);
      |                                                              ^

In file included from /opt/ros/foxy/include/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:22,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/image_streamer.h:4,
                 from /home/ubuntu/dev_ws/src/web_video_server/include/web_video_server/ros_compressed_streamer.h:5,
                 from /home/ubuntu/dev_ws/src/web_video_server/src/ros_compressed_streamer.cpp:1:
[... cut ...]
make[2]: *** [CMakeFiles/web_video_server.dir/build.make:180: CMakeFiles/web_video_server.dir/src/png_streamers.cpp.o] Error 1
make[2]: *** [CMakeFiles/web_video_server.dir/build.make:167: CMakeFiles/web_video_server.dir/src/jpeg_streamers.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/web_video_server.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< web_video_server [30.0s, exited with code 2]

Summary: 0 packages finished [35.2s]
  1 package failed: web_video_server
  1 package had stderr output: web_video_server
```

APhanSpAARC, 2021-04-25:
I'm also receiving errors when building the ROS2 branch. Anyone found a solution to this?

jgvictores, 2023-01-23:
> I'm also receiving errors when building the ROS2 branch. Anyone found a solution to this?

Yes (as always, sorry for bumping stale issues, but I feel it's better than finding no information at all), please see https://github.com/RobotWebTools/web_video_server/issues/105#issuecomment-1400439009

bjsowa, 2024-10-16:
We dropped support for Foxy as it is EoL (you can still build verion 1.0.1 of this package though). The new version was released for all currently supported distros (Humble, Iron, Jazzy, Rolling).

closed 2024-10-16 by bjsowa