RobotWebTools/web_video_server #114: Does not support TLS

AnthonyPhan, 2021-01-26:
Hi i'm unable to find any information on how to get this plugin to work with TLS so that it is not rejected by most modern browsers.

Hanlonian, 2021-02-09:
Hi @AnthonyPhan ,
I recently ran into this problem for a use case where the video server requests needed to be made using https. My solution was to run a very basic proxy server that redirects https requests to the ROS web video server.
I used [NGINX](http://nginx.org/en/docs/beginners_guide.html) for the the proxy server.
This server configuration worked quite nicely:
```
server {
        listen      10080 ssl;
        listen [::]:10080 ssl;
        server_name your_server.com;

        ssl_certificate        /etc/ssl/certs/your_cert.crt;
        ssl_certificate_key    /etc/ssl/certs/your_cert.key;
        ssl_verify_client      off;

        location / {
                proxy_pass http://0.0.0.0:8080;
        }

}
```
This server listens for https requests on the port defined with the listen statements (10080) and uses the certificates defined by the file paths. All requests are then forwarded to the address given by the proxy pass statement and the request uri remains unchanged.
Example:
`https://localhost:10080/stream?topic=/usb_cam/image_raw` is redirected to `http://0.0.0.0:8080/stream?topic=/usb_cam/image_raw`

Best regards,
Matt

gxomalis, 2024-05-27:
hey! did you figure out how to run web video server with ssl? I could use some help cause im trying to do the same thing.