RobotWebTools/web_video_server #131: get a error

wolegeyun, 2022-11-18:
**Description**
throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'

- Library Version:
- ROS Version:
- Platform / OS:

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

wolegeyun, 2022-11-18:
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument

ingjae, 2023-05-23:
I have same error and died

ingjae, 2023-05-25:
It is not a clean solution, but try setting parameter ''**~ros\_threads** " to "**1**" 

I think It has mutex bug so I tried working  just 1 thread then it doesn't died  
  
 

[image]