RobotWebTools/web_video_server #138: Release for ROS2 Humble and Iron

rjb0026, 2023-09-01:
Are there any plans to provide a release (debians) for ROS2 Humble and Iron?

Is anyone still maintaining this project?

shiva-rightbot, 2023-10-26:
it works it humble..

omerts, 2023-12-03:
@shiva-rightbot He means a debian package you can install with `apt` instead of compiling from source

kylenietfeld, 2024-02-08:
I'm also having problems installing on humble. I try _rosdep install --from-paths ~/ros2_ws/src/ --ignore-src -r -y_ 
and I get the following error:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
web_video_server: Cannot locate rosdep definition for [roslib]

varunkotian, 2024-04-08:
It would be great if there is a package for humble

Danendra10, 2024-06-07:
Still waiting for the web video server to release on humble

bjsowa, 2024-09-30:
I'm fixing some issues with newer versions of ffmpeg libraries. Once that's fixed and I'll have the access to release repository, I'll release this package for all currently supported ROS2 distributions (Humble, Iron, Jazzy, Rolling).

> I'm also having problems installing on humble. I try _rosdep install --from-paths ~/ros2_ws/src/ --ignore-src -r -y_
> and I get the following error:
> ERROR: the following packages/stacks could not [...]

@kylenietfeld You need to use the `ros2` branch

bjsowa, 2024-10-16:
The package has been released and is available in testing repositories for Iron, Jazzy and Rolling. The humble release is held for the upcoming sync and will probably be merged in a couple of days and available on the main repo after the next sync (probably a couple of weeks from now).

closed 2024-10-16 by bjsowa