RobotWebTools/web_video_server #143: Cannot build in Ubuntu 24.04 / ROS 2 Rolling

sea-bass, 2024-06-28:
**Description**

I tried building on my personal laptop that runs Ubuntu 24.04 / Rolling and this does not compile for 2 reasons.

1. There is an include to `<cv_bridge/cv_bridge.h>` which broke when I tried because it's been renamed to `.hpp`. Iron seems to have them both available as a transitional release, but no more beyond that. Easy enough to fix this one.

2. Seems like the version of ffmpeg on 24.04 is incompatible, since in `libavstreamer.cpp`` we are including some deprecated code. Seems associaated with https://stackoverflow.com/questions/74268971/migration-from-ffmpeg-4-3-2-to-5-1-0-alternative-to-avlockop-enum

Full error:

```
Starting >>> web_video_server
--- stderr: web_video_server                            
gmake[2]: *** No rule to make target '/opt/ros/rolling/lib/libfastcdr.so.2.2.1', needed by 'web_video_server'.  Stop.
gmake[2]: *** Waiting for unfinished jobs....
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:11:63: error: use of enum ‘AVLockOp’ without previous declaration
   11 | static int ffmpeg_boost_mutex_lock_manager(void **mutex, enum AVLockOp op)
      |                                                               ^~~~~~~~
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp: In function ‘int web_video_server::ffmpeg_boost_mutex_lock_manager(void**, int)’:
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:18:10: error: ‘AV_LOCK_CREATE’ was not declared in this scope; did you mean ‘AV_LOG_TRACE’?
   18 |     case AV_LOCK_CREATE:
      |          ^~~~~~~~~~~~~~
      |          AV_LOG_TRACE
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:25:10: error: ‘AV_LOCK_OBTAIN’ was not declared in this scope
   25 |     case AV_LOCK_OBTAIN:
      |          ^~~~~~~~~~~~~~
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:31:10: error: ‘AV_LOCK_RELEASE’ was not declared in this scope
   31 |     case AV_LOCK_RELEASE:
      |          ^~~~~~~~~~~~~~~
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:37:10: error: ‘AV_LOCK_DESTROY’ was not declared in this scope
   37 |     case AV_LOCK_DESTROY:
      |          ^~~~~~~~~~~~~~~
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp: In constructor ‘web_video_server::LibavStreamer::LibavStreamer(const async_web_server_cpp::HttpRequest&, async_web_server_cpp::HttpConnectionPtr, rclcpp::Node::SharedPtr, const std::string&, const std::string&, const std::string&)’:
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:65:3: error: ‘av_lockmgr_register’ was not declared in this scope
   65 |   av_lockmgr_register(&ffmpeg_boost_mutex_lock_manager);
      |   ^~~~~~~~~~~~~~~~~~~
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:66:3: error: ‘av_register_all’ was not declared in this scope
   66 |   av_register_all();
      |   ^~~~~~~~~~~~~~~
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp: In member function ‘virtual void web_video_server::LibavStreamer::initialize(const cv::Mat&)’:
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:114:35: error: invalid conversion from ‘const AVOutputFormat*’ to ‘AVOutputFormat*’ [-fpermissive]
  114 |   output_format_ = av_guess_format(format_name_.c_str(), NULL, NULL);
      |                    ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                   |
      |                                   const AVOutputFormat*
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:142:34: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  142 |     codec_ = avcodec_find_encoder(output_format_->video_codec);
      |              ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                  |
      |                                  const AVCodec*
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:144:42: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  144 |     codec_ = avcodec_find_encoder_by_name(codec_name_.c_str());
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
      |                                          |
      |                                          const AVCodec*
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:160:35: error: ‘AVStream’ {aka ‘struct AVStream’} has no member named ‘codec’
  160 |   codec_context_ = video_stream_->codec;
      |                                   ^~~~~
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:163:3: error: ‘avcodec_get_context_defaults3’ was not declared in this scope
  163 |   avcodec_get_context_defaults3(codec_context_, codec_);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:189:30: error: ‘CODEC_FLAG_GLOBAL_HEADER’ was not declared in this scope; did you mean ‘AV_CODEC_FLAG_GLOBAL_HEADER’?
  189 |     codec_context_->flags |= CODEC_FLAG_GLOBAL_HEADER;
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~
      |                              AV_CODEC_FLAG_GLOBAL_HEADER
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp: In member function ‘virtual void web_video_server::LibavStreamer::sendImage(const cv::Mat&, const rclcpp::Time&)’:
/home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:294:17: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
  294 |   av_init_packet(&pkt);
      |   ~~~~~~~~~~~~~~^~~~~~
In file included from /usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:44,
                 from /home/sebastian/workspace/web_video_server/include/web_video_server/libav_streamer.h:11,
                 from /home/sebastian/workspace/web_video_server/src/libav_streamer.cpp:1:
/usr/include/x86_64-linux-gnu/libavcodec/packet.h:629:6: note: declared here
  629 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/web_video_server.dir/build.make:104: CMakeFiles/web_video_server.dir/src/libav_streamer.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/web_video_server.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< web_video_server [8.05s, exited with code 2]
```

---

- Library Version:
- ROS Version: Rolling
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**
Clone the `ros2` branch on a Ubuntu 22.04 / ROS 2 Rolling system, then `colcon build` the package.

**Expected Behavior**
It builds without error.

**Actual Behavior**
It builds with error.

bjsowa, 2024-09-17:
Check out [my fork](https://github.com/fictionlab/web_video_server) :)

sea-bass, 2024-09-17:
> Check out [my fork](https://github.com/fictionlab/web_video_server) :)

Ooh, please open a PR!

MarcOlivierFecteau, 2024-09-23:
> Check out [my fork](https://github.com/fictionlab/web_video_server) :)

The build finishes without error on Jazzy. Thank you!

bjsowa, 2024-09-30:
> 1. There is an include to `<cv_bridge/cv_bridge.h>` which broke when I tried because it's been renamed to `.hpp`. Iron seems to have them both available as a transitional release, but no more beyond that. Easy enough to fix this one. [ros-perception/vision_opencv@7a47d35](https://github.com/ros-perception/vision_opencv/commit/7a47d35020d7a40f46fb039b0747c4c0626c796a)

@sea-bass I fixed this in #149 . Would appreciate a review.

closed 2024-10-02 by bjsowa