RobotWebTools/web_video_server #147: Offering help with maintaining this repository

bjsowa, 2024-09-18:
This package hasn't been properly maintained in years. As we are using it in our robots, we have to keep a fork of it to create locally distributed deb packages. I would much rather just support the upstream repo. That's why I am offering my help with maintaining it. This includes:

- releasing the package on active ROS2 distros (Humble, Iron, Jazzy, Rolling)
- adding ament_lint tests
- adding GitHub CI that builds the package and run tests
- sorting out licensing stuff (adding license headers, updating copyright notices)
- Merging or requesting changes on various pull requests
- Closing/Resolving various issues.
- Dropping old, deprecated code
- Changing default branch to `rolling` or `ros2`
- Moving the documentation from ROS wiki to `docs.ros.org` and updating it.

I'm pinging @EzraBrooks as the last self-proclaimed maintainer and @sea-bass as the latest contributor.

EzraBrooks, 2024-09-18:
I'm not sure I have the authority to make this call - I believe that's @amacneil

bjsowa, 2024-09-27:
@amacneil Could you please consider?

amacneil, 2024-09-29:
I've never been involved with this project. But I appear to have access to it, and no one else seems to object, so I sent you an invite 🤷

bjsowa, 2024-09-29:
> I've never been involved with this project. But I appear to have access to it, and no one else seems to object, so I sent you an invite 🤷

Thank you! One more thing is I will need access to the release repository. This can be done by adding me to [robotwebtools](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/robotwebtools.tf) team in `ros2-gbp`. Alternatively, I can add `web_video_server-release` repository to [my team](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/fictionlab.tf) (I'm not sure 2 teams can share access to one repository). Let me know which option do you prefer.

Also, could you change the default branch to `ros2`? I don't have sufficient permissions to do this.

amacneil, 2024-09-29:
> Also, could you change the default branch to ros2? I don't have sufficient permissions to do this.

Done

> One more thing is I will need access to the release repository.

I don't have access to that repo sorry, you will need to make a PR

bjsowa, 2024-09-29:
> I don't have access to that repo sorry, you will need to make a PR

You are listed in the [robotwebtools maintainer team](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/robotwebtools.tf#L4) so you do have access (at least for pushing commits). You can add me to the team by creating [issue in this repo](https://github.com/ros2-gbp/ros2-gbp-github-org/issues/new?assignees=&labels=&projects=&template=update_release_team_membership.md&title=Update+release+team+membership). Doing so will also give me access to release repository for `rosbridge_suite`. If you would rather avoid that, we can try to work around this.

amacneil, 2024-10-01:
Looks like that worked for you.

I also removed myself from the release repo in https://github.com/ros2-gbp/ros2-gbp-github-org/pull/622

closed 2024-10-16 by bjsowa