RobotWebTools/web_video_server #169: Package resolved by rosdep, but not by apt

Ewoodss, 2024-10-28:
**Description**
Don't know if this is the right place too post this, but the package isn't available in the ros2 humble apt repo.

- Library Version: ?
- ROS Version: Humble
- Platform / OS: Ubuntu 22.04

**logs**
```bash
sudo apt update
Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 https://developer.download.nvidia.com/compute/cuda/repos/wsl-ubuntu/x86_64  InRelease                                                                       
Hit:3 https://deb.nodesource.com/node_20.x nodistro InRelease                  
Hit:4 http://archive.ubuntu.com/ubuntu jammy InRelease                         
Hit:5 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:6 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:7 http://packages.ros.org/ros2/ubuntu jammy InRelease
Reading package lists... Done      
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

rosdep install -i --from-path ./ros2_ws/src --rosdistro humble -y -r
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
vision_package: Cannot locate rosdep definition for [tracker]
motor_controls: Cannot locate rosdep definition for [nanolib]
data_gathering: Cannot locate rosdep definition for [OpenCV]
Continuing to install resolvable dependencies...
executing command [sudo -H apt-get install -y ros-humble-web-video-server]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-web-video-server
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-humble-web-video-server] failed
  apt: Failed to detect successful installation of [ros-humble-web-video-server]
```

**Steps To Reproduce**

Install ros2 humble on ubuntu 22.04
add web_video_server too a package.xml
run
```bash
 rosdep install -i --from-path ./ros2_ws/src --rosdistro humble -y -r
```

**Expected Behavior**
it installs this package

**Actual Behavior**
it complains that it can't find the package
error:
Unable to locate package ros-humble-web-video-server

bjsowa, 2024-10-28:
The package is available in the [testing repository](https://docs.ros.org/en/humble/Installation/Testing.html). It will be available in the main repo after the [next sync](https://discourse.ros.org/c/release/humble/73)

bjsowa, 2024-11-10:
There was Humble sync a couple a days ago so the package should be installable by apt now. Only the Rolling distribution is missing now.

bjsowa, 2024-11-28:
Rolling got a new sync yesterday, so the package is finally available in all supported distribtions.

closed 2024-11-28 by bjsowa