RobotWebTools/web_video_server #188: Errors in build on ROS2 Jazzy

jfrancis71, 2025-08-13:
I try to build this package on ROS2 Jazzy using RoboStack's conda environment on Ubuntu 22.04.1.

There is no conda package for this, so I try to build myself.

There is also no conda package for async_web_server_cpp, so I also try to build this.

I clone this repository: https://github.com/fkie/async_web_server_cpp.git and build.

I get errors in file src/http_reply.cpp
There are 4 references to leaf() and I change to filename() (A google search has suggested leaf is deprecated).
Also in include/async_web_server_cpp/http_connection.hpp I add #include <boost/array.hpp> to the list of included files.

I then rebuild. It builds great.

I then try to build web_video_server. It fails.
I alter src/libav_streamer.cpp
Changing:
    static int dispatch_output_packet(void * opaque, uint8_t * buffer, int buffer_size)
    to:
    static int dispatch_output_packet(void * opaque, const uint8_t * buffer, int buffer_size)

and rebuild.

It builds, I test it, it is working and I can now stream HTTP video. Great.

I'd like to ask RoboStack to add this as a conda package, but first I'd like to understand what went wrong with my build process.

Is https://github.com/fkie/async_web_server_cpp.git the correct source for this package? (The last commit was 6 years ago).

Thanks for any help.

bjsowa, 2025-09-02:
@jfrancis71 What do I need to do to include web_video_server to RoboStack distribution? Should I just release the new version to rosdistro?

roehling, 2025-09-04:
I just released version 2.0.1 for [humble](https://github.com/ros/rosdistro/pull/47499), [jazzy](https://github.com/ros/rosdistro/pull/47500), [kilted](https://github.com/ros/rosdistro/pull/47501), and [rolling](https://github.com/ros/rosdistro/pull/47502).

jfrancis71, 2025-09-09:
I have installed and tested web-video-server using RoboStack Jazzy (mamba install) on both a Dell Intel, and Raspberry Pi 3B+ (all Ubuntu 22.04.5)

I am pleased to say it works on both platforms.

Thank you all for your help.

closed 2025-09-09 by jfrancis71